micropython/micropython #17970: ESP32 C6 Micropython I2S Support

kuyeon, 2025-08-22:
### Description

I’m trying to use the MAX98357A with I2S,
but I confirmed that it’s not yet supported.
When will it be supported?
Running from machine import I2S throws an import error.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dpgeorge, 2025-08-22:
See PR #17644 which enables I2S on C6.

kuyeon, 2025-08-22:
Thank you @ricksorensen @dpgeorge I'll try that !

kuyeon, 2025-08-25:
@dpgeorge @Josverl Does that mean it still cannot be used? Is there a way for me to manually adjust the partition size in MicroPython and port it myself?

kuyeon, 2025-08-25:
# Optimise using -Os to reduce size
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_OPTIMIZATION_PERF=n
CONFIG_COMPILER_OPTIMIZATION_CHECKS_SILENT=y

# Change maximum log level to error, to reduce firmware size.
CONFIG_LOG_MAXIMUM_LEVEL_ERROR=y
CONFIG_LOG_MAXIMUM_LEVEL_INFO=n

# Disable SPI Ethernet driver to reduce firmware size.
CONFIG_ETH_USE_SPI_ETHERNET=n

CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions-2MiB.csv"

I am checking this setting. Could you guide me on how to change it?

robert-hh, 2025-08-25:
Just change the respective file esp32/boards/sdkconfig.c6 by adding these lines (either the first or second set from my post) and re-compile.

kuyeon, 2025-08-25:
`from machine import I2S`

works !

and firmware.bin is only 1,784KB, original is 1,949KB