micropython/micropython #17976: Setting hostname on Pico W (still) has no effect: no host name is exposed (also not the default one)

magnus-0, 2025-08-22:
### Port, board and/or hardware

Pico W

### MicroPython version

MicroPython v1.26.0 on 2025-08-09; Raspberry Pi Pico W with RP2040

### Reproduction

1. Create a WLAN with IF_AP and set hostname via network.hostname('test') or via deprecated WLAN.config(hostname = 'test')
2. Confirm that the hostname is printed as such
3. Conclude that **no** hostname is exposed when connected to the Pico W, not even the default one (CYW43)

### Expected behaviour

Expected the hostname to be exposed as set (or at least as default)

### Observed behaviour

No hostname is being exposed (not even the default one)

### Additional Information

According to [this issue](https://github.com/micropython/micropython/issues/10397) the problem was solved and closed, but that doesn't appear to be the case in reality.
Another case is [here](https://github.com/micropython/micropython/issues/8906) - same issue.

There is some writing of people re-building the firmware themselves whilst backing the hostname in it, but that's not a true solution for general Pico users.

See also [this](https://stackoverflow.com/questions/79242474/i-cannot-change-the-hostname-on-a-picow-using-micropython) url.

### Code of Conduct

Yes, I agree

osarhan, 2025-08-29:
yeah similar issue here.  Debating to use an ESP32, since I cant get around this hostname issue... Or figure out how to create a custom build micropython flash file that enables the hostname capability.

osarhan, 2025-08-29:
I found the default hostname to be `picow` and you can access it using `picow.lan`

Not the best solution if you have multiple pico's on your network.. but better than nothing..

magnus-0, 2025-09-01:
I am referring to AP mode indeed, see above.
Good question on where/how the host name would be registered; I dont have information/knowledge of that myself unfortunately.

@osarhan : do you mind sending me a PM with a step-by-step on how you re-build the firmware for that? Its a bit off-topic I guess to post it here, but could be useful for others to read at the same time.

SteveSpencer, 2025-09-02:
In "MicroPython v1.26.0 on 2025-08-09; Raspberry Pi Pico W with RP2040" on a PicoW,
in IF_STA mode, I use:

network.hostname('myhost')
before getting the WLAN up and running. My DNS and DHCP services for the wireless network are providing by a Pi running dnsmasq, and this works fine. I don't manually provide a dhcp-host definition for the mac address, and other machines on the network can resolve and locate the Pico W with no problem, so for WLAN/IF_STA on the Pico W (and the Pico2 W) it works as I expect it to.
(Sadly, the same can't be said for the Wiznet W5100S-EVB-PICO, but that's not _this_ issue.)

osarhan, 2025-09-02:
> I am referring to AP mode indeed, see above. Good question on where/how the host name would be registered; I dont have information/knowledge of that myself unfortunately. [...]

For AP Mode see this [project](https://github.com/simonprickett/phewap) as an example.

For STA mode I used [this section](https://github.com/cbrand/micropython-mdns) of the read me in micropython-mdns.

In the dockerfile I added this to the top:
`ARG HOSTNAME=<YOUR_HOST_NAME>`

and after `RUN sed` statement I added:
`RUN sed -i -E 's/define MICROPY_PY_NETWORK_HOSTNAME_DEFAULT([ ]+)"[^"]+"/define MICROPY_PY_NETWORK_HOSTNAME_DEFAULT\1"'"${HOSTNAME}"'"/g' ports/rp2/boards/RPI_PICO_W/mpconfigboard.h`

It only worked on the 1.24 docker file.. and I was happy with that.