micropython/micropython #18023: ESP32 neopixel library not working properly

kdschlosser, 2025-09-01:
### Port, board and/or hardware

ESP32

### MicroPython version

V1.25

### Reproduction

compile for the ESP32 port using the generic ESP32 board setting the board variant to SPIRAM.attach neopixels to ESP32 and start up the neopixel library and simply fill the buffer with a single color and see if the colors of the neopixels get set to what you have set them to in code. Then repeat the same test except don't set the board variant at all. 

### Expected behaviour

I expect the neopixel colors to be set correctly no matter what the board variant is set to. 

### Observed behaviour

If I compile MicroPython setting the board variant to SPIRAM the neopixel library doesn't set the colors correctly when using the neopixel library. if I don't use the board variant then it works fine.

### Additional Information

I believe the issue is actually upstream in the ESP-IDF but the problem there is MicroPython is using deprecated code in the ESP-IDF for the RMT. I believe that updating the code in the machine_bitstream.c source file to use the new version of the RMT in the ESP-IDF will more than likely solve the issue.

I am working on doing that now and I will let you know the outcome. 

### Code of Conduct

Yes, I agree

ricksorensen, 2025-09-02:
See pr #16293

kdschlosser, 2025-09-02:
and that PR has been around for what? 9 months and has not been merged yet. so I don't foresee it getting merged so the problem still exists. There have been no code changes to the bitstream library that would fix the issue.

ricksorensen, 2025-09-02:
Yes the PR has been around for a while, which most probably indicates it is not thoroughly vetted and not in high demand.

This PR does use the new 5.4x RMT API.  The `machine_bitstream.c` defines `machine_bitstream_high_low_bitbang` which is not used, and `machine_bitstream_high_low_rmt` which uses the new API.
The public `machine_bitstream_high_low` only uses the RMT implementation, and this is the function called by the micropython_lib `neopixel` module.

Additionally the PR modifies the `esp32_rmt.c` file to use new 5.4x API and removes the option to set RMT channel/select SW bitbang,

I will try your experiments when I get back to my desk.

closed 2025-12-01 by kdschlosser