micropython/micropython #18084: ESP32S2 fatal error when running help(machine)

charkster, 2025-09-16:
### Port, board and/or hardware

ESP32-S2-DEVKITM-1U-N4R2 and ESP32-S2-DEVKIT-1-N4R2 boards

### MicroPython version

MicroPython v1.26.1 on 2025-09-11; Generic ESP32S2 module with ESP32S2

### Reproduction

import machine
help(machine)

### Expected behaviour

object <module 'machine'> is of type module
  __name__ -- machine
  mem8 -- <8-bit memory>
  mem16 -- <16-bit memory>
  mem32 -- <32-bit memory>
  unique_id -- <function>
  soft_reset -- <function>
  bootloader -- <function>
  reset -- <function>
  reset_cause -- <function>
  idle -- <function>
  freq -- <function>
  lightsleep -- <function>
  deepsleep -- <function>
  disable_irq -- <function>
  enable_irq -- <function>
  bitstream -- <function>
  dht_readinto -- <function>
  time_pulse_us -- <function>
  Signal -- <class 'Signal'>
  SoftI2C -- <class 'SoftI2C'>
  SoftSPI -- <class 'SoftSPI'>
  ADC -- <class 'ADC'>
  ADCBlock -- <class 'ADCBlock'>
  DAC -- <class 'DAC'>
  I2C -- <class 'I2C'>
  I2S -- <class 'I2S'>
  PWM -- <class 'PWM'>
  SPI -- <class 'SPI'>
  UART -- <class 'UART'>
  WDT -- <class 'WDT'>
  sleep -- <function>
  Timer -- <class 'Timer'>
  Pin -- <class 'Pin'>
  TouchPad -- <class 'TouchPad'>
  RTC -- <class 'RTC'>
  SLEEP -- 2
  DEEPSLEEP -- 4
  HARD_RESET -- 2
  PWRON_RESET -- 1
  WDT_RESET -- 3
  DEEPSLEEP_RESET -- 4
  SOFT_RESET -- 5
  wake_reason -- <function>
  PIN_WAKE -- 2
  EXT0_WAKE -- 2
  EXT1_WAKE -- 3
  TIMER_WAKE -- 4
  TOUCHPAD_WAKE -- 5
  ULP_WAKE -- 6

### Observed behaviour

object <module 'machine' from 'machine.py'> is of type module
  const -- <function>
  Counter -- <class 'Counter'>
  __name__ -- machine
  __file__ -- machine.py
  __getattr__ -- <function __getattr__ at 0x3f802360>
  
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.26.1 on 2025-09-11
IDF version : v5.4.2
Machine     : Generic ESP32S2 module with ESP32S2

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400a8b0b  PS      : 0x00060630  A0      : 0x800a8b53  A1      : 0x3ffe1c30  
A2      : 0x3f02a2a4  A3      : 0x00000004  A4      : 0x00000000  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffe1c00  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x400e4064  A13     : 0x3ffe1c20  
A14     : 0x00000002  A15     : 0x00000310  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400e4064  LEND    : 0x3ffe1c20  LCOUNT  : 0x40029c14  

Backtrace: 0x400a8b08:0x3ffe1c30 0x400a8b50:0x3ffe1c50 0x400e675b:0x3ffe1c70 0x400ab8cb:0x3ffe1ca0 0x400b31c1:0x3ffe1cd0 0x400293e9:0x3ffe1cf0 0x400ab863:0x3ffe1d90 0x400b31c1:0x3ffe1df0 0x400b31d6:0x3ffe1e10 0x400c1e86:0x3ffe1e30 0x400c1fc8:0x3ffe1ec0 0x400a31ca:0x3ffe1f10

ELF file SHA256: 17317db72

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400257fa
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe5110,len:0xe0c
load:0x4004a000,len:0x4
load:0x4004a004,len:0x8ec
load:0x4004e000,len:0x3100
entry 0x4004a134
MicroPython v1.26.1 on 2025-09-11; Generic ESP32S2 module with ESP32S2
Type "help()" for more information.

### Additional Information

Last good release: MicroPython v1.24.1 on 2024-11-29.
All later releases show the bug behavior.

Not sure if this is related:
https://github.com/micropython/micropython/pull/16168

### Code of Conduct

Yes, I agree

ricksorensen, 2025-09-16:
I note that on C3 and S3 GENERIC builds, the module `machine` shows up twice in the `help("modules")` list.  Is this part of the issue?

```
$ mpremote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell

>>> 
MPY: soft reboot
MicroPython v1.26.0 on 2025-08-09; ESP32C3 module with ESP32C3
Type "help()" for more information.
>>> help("modules")
__main__          btree             io                ssl
_asyncio          builtins          json              struct
_boot             cmath             machine           sys
_espnow           collections       machine           time
_onewire          cryptolib         math              tls
_thread           deflate           micropython       uasyncio
_webrepl          dht               mip/__init__      uctypes
aioespnow         ds18x20           neopixel          umqtt/robust
apa106            errno             network           umqtt/simple
array             esp               ntptime           upysh
asyncio/__init__  esp32             onewire           urequests
asyncio/core      espnow            os                vfs
asyncio/event     flashbdev         platform          webrepl
asyncio/funcs     framebuf          random            webrepl_setup
asyncio/lock      gc                re                websocket
asyncio/stream    hashlib           requests/__init__
binascii          heapq             select
bluetooth         inisetup          socket
Plus any modules on the filesystem
>>> 
```

robert-hh, 2025-09-16:
Maybe this is caused by machine being a built-i module and a machine.py module in frozen bytecode, which is the wrapper of machine.PCNT for machine.encoder() and  machine.counter

ricksorensen, 2025-09-16:
Thanks @robert-hh.  That explains the duplicate in `help("modules")`.  I simply renamed the `machine.py` in the `modules`  directory to `nmachine.py` recompiled and verified the original and the new modules are in `help("modules")`.  However, the original error shows up with `import nmachine;help(nmachine)`